EtchedPixels/FUZIX #1088: netd applications not building on Z80

electrified, 2024-09-21:
Hey there.

I see the netd applications are no longer built for Z80.

After modifying Makefile.z80 so they are built (`BUSTEDAPPS` -> `APPS`) I got the following error.

`netd.c:973 - out of index memory`.

Any ideas on what to change/ how much work it would be to get netd working again?

Thanks.

EtchedPixels, 2024-09-24:
That's coming from the compiler running out of internal space rather than a netd problem.

Which config are you building ?

electrified, 2024-09-24:
I'm building `rcbus-z180` - these are the only changes from master

```
diff --git a/Applications/netd/Makefile.z80 b/Applications/netd/Makefile.z80
index 4ee149fae..f44d43dd3 100644
--- a/Applications/netd/Makefile.z80
+++ b/Applications/netd/Makefile.z80
@@ -8,7 +8,7 @@ SRCS += htget.c tinyirc.c linein.c ifconfig.c
 
 OBJS = $(SRCS:.c=.o)
 
-BUSTEDAPPS = netd
+APPS = netd
 
 all: $(APPS)
 
diff --git a/Makefile b/Makefile
index ab6da4394..60e8d9d28 100644
--- a/Makefile
+++ b/Makefile
@@ -89,7 +89,7 @@
 # zxdiv:       ZX Spectrum 128K with DivIDE/DivMMC interface
 # zxuno:       ZX Uno FPGA system
 
-TARGET ?= zrc
+TARGET ?= rcbus-z180
 
 include version.mk
```

EtchedPixels, 2024-09-25:
Should work now if you update kernel and compiler. Turns out much easier to fix than I thought.

electrified, 2024-09-25:
All working now - many thanks.

closed 2024-09-25 by electrified