EtchedPixels/FUZIX #1090: ESP8266 -  A fatal error occurred: No segment header found at offset 0000 in ELF file.

ermiah, 2024-09-26:
I am trying to run FUZIX on an ESP8266 device. I have found multiple problems with the build process so far. 
1.  Running `make TARGET=esp8266 kernel diskimage` gives this error: `drivewire.h:125 file not found`. 
Commenting out the following line in `Library/include/fuzix-include.pkg` resolves this:
`f 0644 /usr/include/sys/drivewire.h sys/drivewire.h`

2) Running `make -C Kernel/platform/platform-esp8266 burn` fails. `platform-esp8266` directory has been relocated to `Kernel/platform`, but the `update-flash.sh` file inside that diriectory has not been updated. 

To resolve this, I have replaced all `../..`s in `update-flash.sh` with `../../../`

3) `make -C Kernel/platform/platform-esp8266 fburn` results in the following error: `A fatal error occurred: No segment header found at offset 0000 in ELF file.` I don't have any idea how to resolve this one. 

A more complete log of this command: 
```
make: Entering directory '/home/**/FUZIX/Kernel/platform/platform-esp8266'
./update-flash.sh
Making 512 byte/block filesystem with normal byte order on device filesystem.img with fsize = 2590 and isize = 32.
Source file '../../../Applications/util/blkdiscard' not found
chmod: can't stat file 'blkdiscard': 2
# -s is the flash size in kilobytes; the kernel and boot loader takes the first 64kB.
../../../Standalone/mkftl -s 1984 -e 0x1000 -p 512 -g 10 filesystem.img -o filesystem.ftl
Number of physical erase blocks: 496
Maximum logical size: 1299kB (2598 sectors)
Used space: 1295kB (2590 sectors)
xtensa-lx106-elf-objcopy -I binary -O elf32-xtensa-le --change-section-vma .data=0x40220000 filesystem.ftl filesystem.elf
esptool elf2image filesystem.elf
esptool.py v4.8.0

A fatal error occurred: No segment header found at offset 0000 in ELF file.
make: *** [Makefile:86: fburn] Error 2
make: Leaving directory '/home/**/FUZIX/Kernel/platform/platform-esp8266'
```

EtchedPixels, 2024-09-26:
Source file '../../../Applications/util/blkdiscard' looks the relevant bit

EtchedPixels, 2024-09-26:
Seems right for make burn now but not fburn. Can't test on actual hardware as my stuff is all packed away

closed 2024-09-26 by EtchedPixels