EtchedPixels/FUZIX #1096: Failed to build rcbus-z180

wdl1908, 2024-10-12:
make[2]: Entering directory '/src/FUZIX/Applications/util'
fcc -mz80 -X -Os -D__STDC__ -c -I/src/FUZIX/Library/include -I/src/FUZIX/Library/include/z80   -c -o banner.o banner.c
<command-line>:0:0: warning: "__STDC__" redefined
<built-in>: note: this is the location of the previous definition
/src/FUZIX/Library/link/ldz80 /src/FUZIX/Library/libs/crt0_z80.o banner.o -o banner -L/src/FUZIX/Library/libs -lcz80
/src/FUZIX/Library/libs/libcz80.a: Unknown symbol '_close'.
Makefile.common:130: recipe for target 'banner' failed

EtchedPixels, 2024-10-17:
Can you attach the full build log - that failed because the libc is messed up so the real fail must be earlier.

wdl1908, 2024-10-17:
[make.log](https://github.com/user-attachments/files/17422793/make.log)

wdl1908, 2024-10-17:
hmm wrong log the previous was for make TARGET=sc720
this one is for make TARGET=rcbus-z180
[make.log](https://github.com/user-attachments/files/17422880/make.log)

EtchedPixels, 2024-10-20:
join: 29133symref:146: is not sorted: __bool fmaxf.o
ln -sf mz80.lib libmz80.a

join: 28487symref:9: is not sorted: __cmpeq readline.o

What version of "join" are you using and what are you building on ?

wdl1908, 2024-10-20:
$ join --version
join (GNU coreutils) 8.26
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Mike Haertel.

It's a Debian 9 system

EtchedPixels, 2024-10-21:
What is failing is the library sorting code

```
sort $$symdef -o $$symdef
sort $$symref -o $$symref
join $$symref $$symdef | sed 's/[^ ]* *//'
```

and I'm a bit at loss how after they've been sorted this can break (especially as that particular piece of code goes back to the 1970s)

Can't reproduce it here at the moment either.

closed 2024-12-17 by EtchedPixels