EtchedPixels/FUZIX #1166: Fuzix/GUI/UTK

alex-arknetworx, 2025-03-16:
Hello @EtchedPixels,

I was interested to know where the GUI toolkit portion of Fuzix might have been headed to or the inspiration for where it might have started from. 

I have looked in README, UTK and the source code for some clues and from what i can tell an ASCII renderer has been developed so far for testing. Which hardware platform has the development of UTK taken place on to date?

Was the initial goal to add a video console to Fuzix or more of a small tiny "drawterm like" window manager (say like Plan 9's rio or Bellcore ManaGeR)?

Any additional info or insights appreciated.

EtchedPixels, 2025-03-16:
Mostly all I've done so far is sketch out some ideas. The text based renderer simply being because it's easier to play with.

Given how little memory there is on most systems and how little resolution I've been looking more at Apple mousetext and the Apple IIe (and equivalent 8bit setups) level of rendering algorithms and routines.

A big problem is also how to generalize it at all.

So I guess the TLDR is "no idea" 8)

alex-arknetworx, 2025-03-17:
Appreciate the reply @EtchedPixels 

At first i thought when you said MouseText you meant MouseDesk: it amazes me how such a complete looking windowing experience could be written for 8-bit on monochrome with 128k, not sure if you've seen the continued development of it at www.a2desktop.com unfortunately its written in assembly: if only it was available in C to make a port an easier prospect.

Then i got what you meant: box drawing characters and the early inclusion of them in Apple ii to stay in text mode. Smart idea to accelerate UI performance :)

Anyway just wanted to understand where this was at in Fuzix and you've satisfied my curiosity: I've only just built Fuzix for RP2040. I have a bunch of ported windowing systems on my GitHub, so might tinker with Fuzix graphics output and see what transpires in the GUI space.

EtchedPixels, 2025-03-18:
I didn't look at the internals of the Apple one although I spent some time moving windows around and looking at redraw patterns to see how it did redrawing without keeping a lot of expensive damage lists, and then implemented a variant of the same but with small improvements.

There are a bunch of Fuzix platforms that really only have a character display, or that have character sized attributes so it seemed worth looking at text mode as one option. In theory it's an implementation detail in the library for that hardware - one thing I'm fairly sure it does need is runtime loadable libraries, but that's not hard to do.