EtchedPixels/FUZIX #1167: bdread and bdwrite 'invalid dev' kernel panics on Z80-MBC2

SanPollo, 2025-03-26:
I have been trying out the [0.4 release version](https://www.fuzix.org/downloads/0.4/z80-mbc2/) of FUZIX on my Z80-MBC2 for a few days, and after a few minutes of using the system I either get:

```
bdread:
panic: invalid dev
```

or:

```
bdwrite:
panic: invalid dev
```

I've tried four different microSD cards, and two different SD modules, all with the same result. Also CP/M 3 runs fine on the computer, so I don't believe this is a hardware issue.

Unfortunately, I do not currently have a build environment to see if a version built from the repository would fix this issue; I use FreeBSD, and have not been able to build the [Fuzix Compiler Kit](https://github.com/EtchedPixels/Fuzix-Compiler-Kit).

Has anyone else experienced the same issues? If so, were you able to work around them?

EtchedPixels, 2025-03-27:
That's almost certainly failing because of some kind of memory management problem. The buffer cache is getting scribbled on which generally means something flipped the banks and didn't put them back. It only ever worked for me with a specific firmware or two and as it's basically an emulator tacked to a Z80 I never really looked too hard what was going on as it's not really much of an interest.

For the FreeBSD and building please file a bug or whatever with the build fails. I definitely want to get the tools fixed to build on *BSD as well.

SanPollo, 2025-03-31:
Many thanks for the explanation. Even before it panics, it seems incredibly slow, and "free" shows the following immediately after booting:

```
         total          used          free
Mem:        64            64             0
Swap:      512            32           480
```

Incidentally, the only time it doesn't seem to crash is when it runs fsck after a panic. Single user mode maybe?

The ATmega32A does perform a lot of functions for the Z80-MBC2, but that seems to be fairly common with modern 8-bit machines like the Neo6502 and the Agon Light. I just finished building my Z80-MBC2 with uTerm, and I am really enjoying playing around with it. I also tried FUZIX [on my Agon Light](https://github.com/tomm/FUZIX/releases), and was very impressed with the speed, and the RAM it can utilise.

I looking forward to trying out the [CPC version](https://www.cpcwiki.eu/forum/applications/port-of-fuzix-for-the-cpc6128/
) too if I can get the build tools compiled.

I will definitely log an issue for building on FreeBSD. I have also just installed NetBSD on an old laptop to test it there too. I think I might have a better chance with NetBSD as it uses GCC as its default compiler.

EtchedPixels, 2025-04-03:
It's a pretty slow system to run Fuzix on as the I/O performance isn't great and there's not a lot of memory so it has to keep swapping stuff back and forth.
The fact it only crashes in multi-user is a useful clue. Odds are that's task switching during an interrupt (ie pre-emption).

SanPollo, 2025-04-08:
I was quite surprised actually because it runs quite quickly with CP/M+, but that's a much more basic OS of course. Is this something that could be fixed in the firmware, or is it related to the design of the board? Sorry for my ignorance, but my knowledge of the lower levels of computing is still at a fairly rudimentary level.

Do you think the crashing issue could be fixed in Fuzix if it could be tracked down, or is that something that needs to be sorted in firmware too?

I logged a new issue for compiling Fuzix-Compiler-Kit on FreeBSD, as promised. I will also log a couple of issues for the [emulator](https://github.com/ivanizag/z80-mbc2-emu) too as it would be really good to be able to test things in software. At the moment it quits at the boot prompt when trying to start Fuzix. Looking at the computer booting, the HALT light is on, so I guess the emulator sees the halt, and thinks it's finished for the day.

EtchedPixels, 2025-04-09:
I don't yet know if it's firmware or something in the pre-emption logic, maybe a race. The firmware I have is before the interval interrupt stuff was added so that path has never been properly tested.
If it's firmware then it can be worked around but will make it even slower 8(