EtchedPixels/FUZIX #1172: RC2014 platform build issues

hechtus, 2025-04-11:
I'm trying to build an run FUZIX on a RC2014 FPGA platform. I'm running into some issues.

The game 2048 has some fcc issues. I'm using the latest fcc build from github.
```
make[2]: Entering directory '/home/ronald/tmp/Fuzix/FUZIX/Applications/games/2048'
fcc -mz80 -X -Os -D__STDC__ -c -I/home/ronald/tmp/Fuzix/FUZIX/Library/include -I/home/ronald/tmp/Fuzix/FUZIX/Library/include/z80 -Os   -c -o draw.o draw.c
draw.c:175 - bad type

fcc -mz80 -X -Os -D__STDC__ -c -I/home/ronald/tmp/Fuzix/FUZIX/Library/include -I/home/ronald/tmp/Fuzix/FUZIX/Library/include/z80 -Os   -c -o main.o main.c
main.c:19 - invalid type conversion
```

The function `ch375_probe()` seems to be missing.
```
?ASlink-Warning-Undefined Global '_ch375_probe' referenced by module 'discard'
```

After hotfixing these issues the kernel starts but it does not find my CTC @ I/O 0x88
```
Volume "Fuzix WBW Loader" [0xF200-0xF400, entry @ 0xF200]...FUZIX version 0.5
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2025 Alan Cox <alan@etchedpixels.co.uk>
Devboot
RomWBW 3.5 on an RC2014@7MHz.
Warning: no timer available.
tty1: Z80 SIO at 0x0080.
tty2: Z80 SIO at 0x0082.
512KiB total RAM, 400KiB available to processes (20 processes max)
Enabling interrupts ... ok.
0000 :  - absent
0002 :  - absent
sd0: hda: hda1 hda2 (swap) 
Resetting SCSI bus... 
```

As it does not find a suitable timer scsi_reset() does not work as it expects a timer to be working.

I had a look the the CTC detection in `rc2014.s` and I really don't understand how this should work
```
	;
	; Probe for a CTC
	;

	dec c				; CTC 2

	ld a,#0x47			; CTC 2 as counter
	out (c),a
	ld a,#0xAA			; Set a count
	out (c),a
	in a,(c)

	cp #0xAA
	jr z, ctc_maybe
	cp #0xA9			; Could be one less
	ret nz

```
If the CTC input is connected to the CPU clock it would definitely count more than one cycle up.

hechtus, 2025-04-11:
Skipping the `scsi_probe()` allows me to enter a boot device, but unfortunately `/init` is crashing.
```
sd0: hda: hda1 hda2 (swap) 
Resetting SCSI bus... OK
bootdev: 1
Mounting root fs (root_dev=1, ro): OK
Starting /init
3 buffers added
[trap_illegal]

monitor: SP=EFBD, PC=F4A5
:( 

```

EtchedPixels, 2025-04-15:
If the CTC input is connected to the CPU clock then it's not configured as Fuzix expects it. Also I'm not sure it's even valid to do so ?

EtchedPixels, 2025-04-15:
2048 and ch375 bits fixed. Will test when my office is back in shape (been redecorating)

hechtus, 2025-04-16:
The 2048 and ch375 build issues are fixed. Regarding the CTC: The output of channel 3 seems to be fed into channel 2. I did it the other way around. Now the CTC is detected.

Now I'm running into random crashes after `Starting /init`
```
Devboot
RomWBW 3.5 on an RC2014@7MHz.
Z80 CTC detected at 0x88.
tty1: Z80 SIO at 0x0080.
tty2: Z80 SIO at 0x0082.
512KiB total RAM, 400KiB available to processes (20 processes max)
Enabling interrupts ... ok.
0000 :  - absent
0002 :  - absent
sd0: hda: hda1 hda2 (swap) 
bootdev: 1
Mounting root fs (root_dev=1, ro): OK
Starting /init
3 buffers added
[trap_illegal]

monitor: SP=F0F6, PC=F4A5
:( 
```
```
...
Starting /init
3 buffers added
[NMI]

monitor: SP=F0ED, PC=7229
:( 
```
```
Starting /init
3 buffers added
bdwrite:
O�d������O��A�d�[NMI]��A�d�#z��'�����O�XH����O��A�d�#����O��A�d�#z��'�����O�XH�O�XH�O�

monitor: SP=F0ED, PC=F25F
:( 
```

The `[NMI]` crash is strange as the NMI input is always pulled high. Do you have any suggestions on how to track down this issue?

hechtus, 2025-04-16:
I was able to reproduce the kernel panics using the EmulatorKit
```
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2025 Alan Cox <alan@etchedpixels.co.uk>
Devboot
RomWBW 3.5 on an RC2014@7MHz.
Warning: no timer available.
PS/2 self test: FF

tty1: ACIA at 0x00A0.
512KiB total RAM, 400KiB available to processes (20 processes max)
Enabling interrupts ... ok.
0000 :  - absent
0002 :  - absent
sd0: hda: hda1 hda2 (swap) 
Mounting root fs (root_dev=1, ro): OK
Starting /init
4 buffers added

panic: corrupt inode
monitor: SP=F472, PC=579A
:( 
```

hechtus, 2025-04-17:
Using the rc2014 emulator (EmulatorKit) I tried the **IDE** diskimage instead of **SD card** emulation. This seems to work
```
Enabling interrupts ... ok.
0000 : ACME ULTRASONICUS AD INFINITUM v0.1 - OK
0001 :  - absent
hda: hda1 hda2 (swap) 
0002 :  - absent
0003 :  - absent
Mounting root fs (root_dev=1, ro): OK
Starting /init
4 buffers added
init version 0.9.1
Checking root file system.
Current date is Thu 2025-04-17
Enter new date: 
```
So, this seems to be an SD card issue and message `corrupt inode` makes more sense.

As I don't have an IDE on my machine I was starting with SD card and probably switching to CH375.

EtchedPixels, 2025-04-17:
Problem identified. It's indeed a problem with the SD driver that happened to get broken when some bits of code ended up moving bank and thus showing up wrong usage of __z88dk_fastcall

Fixing in progress

hechtus, 2025-04-19:
I tried out your recent commits but I don't experience any improvements. Is it still in progress?

I also tried the CH375 support which is working perfectly in RomWBW but not in FUZIX. I was at least trying to fix the ch375_probe() function with some success (I was getting timeouts related to the command `ch375_wcmd(0x51)`) But unfortunately ` ch375_rblock(dptr)` does not work. Could it be the same fastcall/banking issue as for sd card support?

hechtus, 2025-04-21:
Finally, I got it to run. The functions `rc2014.s:_ch375_rblock` and `rc2014.s:_ch375_wblock` are definitely broken. I replaced them with naked c functions like this
```
void ch375_rblock(uint8_t *p) __naked
{
    __asm
#ifdef CONFIG_BANKED
            pop bc
            pop de
            pop hl
            push hl
            push de
            push bc
#else
            pop de
            pop hl
            push hl
            push de
#endif
            ld a, (_td_raw) 			    ; I/O type ?
            push af
#ifdef SWAPDEV
            cp #2
            jr nz, not_swapin
            ld a, (_td_page)	 		   ; swap page
            call map_for_swap
            jr doread
not_swapin:
#endif
            or a                                    ; test is_user
            call nz, map_proc_always		    ; map user memory first if required
doread:
            ld bc, #(0x4000 | CH375_REG_DATA)       ; setup port number
                                                    ; and count
	    inir
            pop af
            or a
            jp nz, map_kernel_restore               ; else map kernel then return
            ret
    __endasm;
}
```
The `ch375_probe()` code still looks ugly. I know from experience that bringing up the ch375 module is time critical. Thus, I'm not sure how to fix it for all platforms.

For reference, this is my boot log
```
RomWBW 3.5 on an RC2014@7MHz.
Z80 CTC detected at 0x88.
tty1: Z80 SIO at 0x0080.
tty2: Z80 SIO at 0x0082.
512KiB total RAM, 400KiB available to processes (20 processes max)
Enabling interrupts ... ok.
0000 :  - absent
0002 :  - absent
ch375: version 43
ch376: firmware version 3
hda: hda1 hda2 (swap) 
bootdev: 1
Mounting root fs (root_dev=1, ro): OK
Starting /init
4 buffers added
init version 0.9.1
Checking root file system.
Current date is Mon 2025-04-21
Enter new date: 
Current time is 07:56:53
Enter new time: 

 ^ ^
 n n   Fuzix 0.5
 >@<
       Welcome to Fuzix
 m m

login: root

Welcome to FUZIX.
# 
```

hechtus, 2025-04-24:
I can confirm that the SD card support is now working better, but I still get errors or panics:
```
init version 0.9.1
hda: I/O error
hda: I/O error
hda: I/O error

 ^ ^
 n n   Fuzix 0.5
 >@<
       Welcome to Fuzix
 m m

login: root
login: unable to change owner of controlling tty
```
```
Starting /init
4 buffers added
hda: I/O error

panic: killed init
monitor: SP=EFE2, PC=579A
:( 
```
Seems to be tricky. As I was already using the SD card interface for (successfully) accessing a FAT file system I suspect that it's not an hardware issue.

Regarding the CH375 probe, I was able to test it at lower clock frequencies (1 MHz) but at 10 MHz it doesn't work. So, my suggestion seems to be error prone. If I'm getting some new results I will let you know.