EtchedPixels/FUZIX #1173: ZX DivMMC platform seems to be broken

hechtus, 2025-04-19:
I'm just trying out the zxdiv platform in fuse emulator and on an FPGA emulator. Both show the same issue. After booting into the `bootdev:` question weird pixel pattern show up. When entering a boot device id or forcing it to scroll up by pressing the space key the system hangs. Looks like as if data is accessed in the wrong page or memory location.

[image]

hechtus, 2025-04-29:
The reason for this is, that the _DATA section overlaps with the screen memory at 0x4000. `fuzix.map` shows
```
...
     00003F98  _lastname                          filesys
     00004010  _ttydata                           tty
     00004065  _vtattr                            vt
     00004066  _vtpaper                           vt
     0000406C  _swappage                          swap
     0000407E  _keyboard_grab                     devinput
     0000410B  _vtborder                          devtty
     0000410C  _cursorpos                         devtty
     0000410E  _ide_unit                          tinyide
     00004112  _sd_shift                          tinysd
     00004114  _tinysd_busy                       tinysd
     00004115  _tinysd_unit                       tinysd
     0000411A  _td_page                           tinydisk
     0000411B  _td_raw                            tinydisk
     0000411C  _td_lba                            tinydisk
     00004144  _td_unit                           tinydisk
     00004146  _td_op                             tinydisk
     0000414A  _td_iop                            tinydisk
     0000414E  _td_next                           tinydisk_discard
     00004175  _fuller                            devinput
     00004176  _kempston                          devinput
     00004177  _kmouse                            devinput
     00004178  _keybuf                            zxkeyboard
     00004180  _keymap                            zxkeyboard
```
I added `#define CONFIG_SMALL` to config.h to reduce the size of the common data sections. This allows me to successfully boot FUZIX.

EtchedPixels, 2025-04-30:
Thanks - added to config.h in the tree.

closed 2025-04-30 by EtchedPixels